raspberrypilearning/balloon-pi-tay-popper #3: Remove lightly coloured wires

gbaman, 2015-02-15:
Used this activity at the Northern Ireland Raspberry Jam today. All our worksheets are printed out black and white half way through the activity I noticed, the yellow wire on the breadboard goes **near completely invisible**.
Caused quite a few issues for the kids doing the activity.

Would be an excellent idea to stay clear of any lightly coloured wires on a white breadboard in the wiring diagrams, as I am sure I won't be the only one printing the activity out in black and white.

LizUpton, 2015-02-15:
Thanks @gbaman. Sam's on vacation this week, but I'll talk to him about it next week: it may be worth outlining paler wires in black for this reason. Thanks very much for the spot!

bennuttall, 2015-06-19:
Thanks - will bear in mind.

Closing as this is issue is not specific to this resource.

closed 2015-06-19 by bennuttall