raspberrypilearning/balloon-pi-tay-popper #5: Speaker wire not needed

gbaman, 2015-02-15:
At today's NI Raspberry Jam, we just used normal single core wire for the activity as we weren't able to cheaply (speaker wire is pretty expensive) source decent amount of speaker wire. The activity worked perfectly fine. 
It easily allowed more than enough current to pop the balloons.
Any reason speaker wire is recommended? May at least be worth mentioning normal wire can be used as well in the parts list for those without decent budget?
It is £12 per balloon wired up with the Maplin sourced speaker wire.

bennuttall, 2015-02-24:
We thought the speaker wire was cheaper, but you're right, we'll change it to core wire.

Can you recommend a place to buy? A link would be handy :)

gbaman, 2015-02-24:
An excellent question.
Pimoroni stick a hookup wire kit - http://shop.pimoroni.com/products/hook-up-wire-spool-set
Although it is a little pricey. Might be able to get away with this, not certain though - http://www.maplin.co.uk/p/bell-cable-single-core-priced-per-metre-xs92a

bennuttall, 2015-02-24:
That looks better (I'm sure we had that in at one point).

So Maplin's [bell cable](http://www.maplin.co.uk/p/bell-cable-single-core-priced-per-metre-xs92a) is 69p per metre compared to their [speaker wire](http://www.maplin.co.uk/p/speaker-cable-figure-of-8-6-x-3mm-white-priced-per-metre-xr60q) at £1.59 per metre.

@RZRZR sound good? Shall I update it?

RZRZR, 2015-02-24:
The pimoroni wire would work out about £11.88 worth for three balloon too - because you'd need twice as much as it's single core.

We went for bell wire because it's easy to pick up at any length, has nice thick core and is conveniently 2-core. Yes @bennuttall, it was bell cable originally - must have got lost in the confusion - let's change it back.

commit 4b9ac9e2 (2015-02-24): Replace speaker wire with bell wire, close #5
closed 2015-02-24 by bennuttall