raspberrypilearning/balloon-pi-tay-popper #9: doesn't work for older Pi

juggledad, 2015-03-14:
I've got a Pi (2011) and these instructions do not work!!! Turns out that where this is using GPIO 2 for the base on the transistor if a noob (like me) follows the pictures, they end up actually connecting the  base to GPIO 0 (SDA) and it is always on! 

I was finally able to get it to work by using GPIO 17 and changing the code.

BUT imagine a kid trying to use this - or a teacher wanting to use this in a class and it won't work - or worse, they have a mix of PI's some new, some old. On some PI's the project works and on others it doesn't...how fustrating would that be???

You need someone to go thru these MAKE's and make sure they work for all the PI's.

juggledad, 2015-03-14:
In addition, you should have a set of alternative parts. I could not buy the transistor specified and used a 2n2222 instead which caused it's own chalange since the leads are in a different order from the BC635.

MarcScott, 2016-06-14:
Resource has now been updated and should be clearer.

closed 2016-06-14 by MarcScott