raspberrypilearning/balloon-pi-tay-popper #10: Balloon pops without code

brijrajsingh, 2015-11-01:
Please see my attached diagram, I am using your code and identical connections, the "pop" msg printing with tactile switch works just fine, but when i do the full implementation, i find that the balloon pops the moment i input the battery.
I later started checking with LED, there is no effect over the LED, neither it gets hot or turns off by the code given, the code doesn't show any error as well

I however see this warning in my console

balloon_updated.py:13: RuntimeWarning: A physical pull up resistor is fitted on this channel!
  GPIO.setup(balloon, GPIO.OUT)
[image]
[image]
[image]

juggledad, 2015-11-01:
Make sure you transistor is in the correct way. Some transistors have the EBC in a different order.

brijrajsingh, 2015-11-02:
Should I just revert it in such a case, So if its the other way around it should work, I just hope I don't end up burning my pi.

juggledad, 2015-11-02:
You need to look at the data sheet for the transistor from the company that makes it (google it) to see the correct layout.

brijrajsingh, 2015-11-04:
yes, Found that my transistor had order EBC unlike yours, the code works like a charm

closed 2015-11-04 by brijrajsingh