raspberrypilearning/balloon-pi-tay-popper #11: copy edit

MarcScott, 2016-06-13:
Hi @lauraclay / @LornaLynch 
Could I get a copy edit on this update please:
New fork is at - https://github.com/raspberrypilearning/balloon-pi-tay-popper/tree/gpiozero-update
Comparison of new and original at - https://github.com/raspberrypilearning/balloon-pi-tay-popper/compare/gpiozero-update?expand=1

I've changed some diagrams and updated the copy and code for gpiozero
Not too many changes

Thanks

LornaLynch, 2016-06-13:
Is it a problem that, in the section where the speaker cable/resistor circuit is being prepared, there's no red or black electrical tape on the cables, though this is mentioned in the worksheet?

MarcScott, 2016-06-13:
It's not a massive issue. I'll take some new photos when I'm back in the office. Good spot though.

LornaLynch, 2016-06-13:
Final one from me, and then the edit is done: I'd say the interactive calculator suggestion in what next might need a tiny bit of explanation? Is there something that could be linked to?

LornaLynch, 2016-06-13:
Other than this I'd say you're ok to close this issue.

MarcScott, 2016-06-13:
Thanks @LornaLynch

MarcScott, 2016-06-13:
@davidhoness was this your resource originally? Could you clarify the interactive calculator suggestion at the end.

bennuttall, 2016-06-13:
It was an external one, rewritten by me & Rachel. I'll take a look.

lauraclay, 2016-06-15:
Reopening, as I've spotted some further edits are needed. (Apologies for just seeing this- problems migrating my email meant I've missed a week of notifications!)

LornaLynch, 2016-06-15:
Apologies @lauraclay :(

closed 2017-02-24 by lawsie